tenstorrent/tt-kmd #15: Installation instructions don't specify that the repo needs to be cloned first

bbradelTT, 2024-04-01:
The instructions at https://github.com/tenstorrent/tt-kmd/blob/main/README.md specify to install dkms via apt or dnf and then to execute the following:
```
sudo dkms add .
sudo dkms install tenstorrent/1.27.1
sudo modprobe tenstorrent
```
However, that may not work. A preliminary step that is needed first is to run
```
git clone https://github.com/tenstorrent/tt-kmd.git
cd tt-kmd 
``` 

It would be good to update the instructions with this information.

alewycky-tenstorrent, 2024-04-01:
The instructions were originally written assuming you had unpacked a tar, but now it's hosted on github which displays the README in the git web UI.

We have to include instructions to check out the current release tag, main is not full tested.

I'm considering what to write and I'll get around to it soon.

warthog9, 2024-04-26:
I'll also note that the README.md's are also generally targeted at more developer type users, not end users, so the assumption of "you need to have cloned this first" while on Github is not a huge stretch.  End users should be out here https://docs.tenstorrent.com/tenstorrent/add-in-boards-and-cooling-kits/grayskull-tm-e75-e150/software-setup with the more end user level documentation.

stalkerg, 2024-05-02:
yeah, here should be `make modules_install` like stuff

warthog9, 2024-05-02:
@stalkerg that should work out of the box right now:

https://github.com/tenstorrent/tt-kmd/blob/e090dd44847af0dbc25b6768fb1daaa44b5a332b/Makefile#L17

though I guess we don't have explicitly noted in the README.md

warthog9, 2024-06-04:
Closing this as it's end users focused docs discussion which should happen outside this repository