tenstorrent/tt-kmd #19: Do you have plans to upstream it?

stalkerg, 2024-05-02:
As I can see device path is unusual for `char` device, but everything else is looks good.

warthog9, 2024-06-04:
Until we've got the driver stabilized, and/or we've got a better idea of where upstream is generally heading with accelerator cards like these, we are going to hold off pushing to get the driver into mainline.

closed 2024-06-04 by warthog9

stalkerg, 2024-06-05:
@warthog9 
1. Kernel has a new class of devices - accelerators. https://docs.kernel.org/accel/introduction.html
2. Based on my experience, if you are not sure yet how to do it, it makes sense to ask questions on LKML; it can prevent a lot of miscommunications and unexpected requirements. 

Regards,

warthog9, 2024-06-05:
This isn't a question of wanting to upstream, trust me you'll find my name in the MAINTAINERS file, this is mostly a we need to stabilize our side first, and see how it plays out, as well as let the new subsystems settle before we figure out how we actually fit in there.

For now: we don't have an immediate plan to push towards mainline as a result of a lot of flux on both sides.