tenstorrent/tt-kmd #32: udev rule assigns insecure file permissions

afaerber, 2024-10-02:
`udev-50-tenstorrent.rules` (referenced by `dkms-post-install`) assigns `MODE="0666"` to the device file.

This is considered insecure, as it gives not just human users but any system services/daemons (including web servers) write access to the hardware.

Better would be to assign `MODE="0660"` and optionally a suitable group. The users who should have access can then get that group assigned individually.

It's trivial to not install this rule in distro packages, but it would probably be good to fix this for `dkms` users, too.

smehtaTT, 2025-01-24:
Nice to have - moving to P2 for now, will bump up to P1 if critical.

joelsmithTT, 2025-10-17:
0666 is default for NVIDIA and is documented by AMD's ROCm stack as a supported alternative to group management for simplifying deployment.  Given that 0666 and 0660 are both established industry practices, forcing a breaking change on our user base and internal teams is not a good tradeoff.