tenstorrent/tt-kmd #47: DKMS module fails to compile on Redhat 8.x

owainkenwayucl, 2025-03-18:
I've been trying to get things set up on one of our RHEL 8 test sytems (a very odd config where the node has some other accelerators in it which require RHEL 8) and apart from some struggles with package names (e.g. `python3` -> `python39`) which I resolved easily, I've run up against some issues compiling the kernel DKMS module.

I'm aware that RHEL 8 is quite old and with kernel 4.18 the DKMS driver fails to compile with:

```
DKMS (dkms-3.1.5) make.log for tenstorrent/1.31 for kernel 4.18.0-477.15.1.el8_8.x86_64 (x86_64)
Tue 18 Mar 15:18:35 GMT 2025
Cleaning build area
# command: make -C /lib/modules/4.18.0-477.15.1.el8_8.x86_64/build M=/var/lib/dkms/tenstorrent/1.31/build clean
make: Entering directory '/usr/src/kernels/4.18.0-477.15.1.el8_8.x86_64'
make: Leaving directory '/usr/src/kernels/4.18.0-477.15.1.el8_8.x86_64'

# exit code: 0
# elapsed time: 00:00:00
Building module(s)
# command: make -j48 KERNELRELEASE=4.18.0-477.15.1.el8_8.x86_64 -C /lib/modules/4.18.0-477.15.1.el8_8.x86_64/build M=/var/lib/dkms/tenstorrent/1.31/build
make: Entering directory '/usr/src/kernels/4.18.0-477.15.1.el8_8.x86_64'
  CC [M]  /var/lib/dkms/tenstorrent/1.31/build/module.o
  CC [M]  /var/lib/dkms/tenstorrent/1.31/build/chardev.o
  CC [M]  /var/lib/dkms/tenstorrent/1.31/build/enumerate.o
  CC [M]  /var/lib/dkms/tenstorrent/1.31/build/interrupt.o
  CC [M]  /var/lib/dkms/tenstorrent/1.31/build/grayskull.o
  CC [M]  /var/lib/dkms/tenstorrent/1.31/build/wormhole.o
  CC [M]  /var/lib/dkms/tenstorrent/1.31/build/blackhole.o
  CC [M]  /var/lib/dkms/tenstorrent/1.31/build/pcie.o
  CC [M]  /var/lib/dkms/tenstorrent/1.31/build/hwmon.o
  CC [M]  /var/lib/dkms/tenstorrent/1.31/build/sg_helpers.o
  CC [M]  /var/lib/dkms/tenstorrent/1.31/build/memory.o
In file included from /var/lib/dkms/tenstorrent/1.31/build/sg_helpers.c:1:
/var/lib/dkms/tenstorrent/1.31/build/sg_helpers.h:17:19: error: redefinition of ‘dma_map_sgtable’
 static inline int dma_map_sgtable(struct device *dev, struct sg_table *dma_mapping, enum dma_data_direction dir, unsigned long attrs)
                   ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/tenstorrent/1.31/build/sg_helpers.h:5,
                 from /var/lib/dkms/tenstorrent/1.31/build/sg_helpers.c:1:
./include/linux/dma-mapping.h:396:19: note: previous definition of ‘dma_map_sgtable’ was here
 static inline int dma_map_sgtable(struct device *dev, struct sg_table *sgt,
                   ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/tenstorrent/1.31/build/sg_helpers.c:1:
/var/lib/dkms/tenstorrent/1.31/build/sg_helpers.h:27:20: error: redefinition of ‘dma_unmap_sgtable’
 static inline void dma_unmap_sgtable(struct device *dev, struct sg_table *dma_mapping, enum dma_data_direction dir, unsigned long attrs)
                    ^~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/tenstorrent/1.31/build/sg_helpers.h:5,
                 from /var/lib/dkms/tenstorrent/1.31/build/sg_helpers.c:1:
./include/linux/dma-mapping.h:419:20: note: previous definition of ‘dma_unmap_sgtable’ was here
 static inline void dma_unmap_sgtable(struct device *dev, struct sg_table *sgt,
                    ^~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/tenstorrent/1.31/build/sg_helpers.c:1:
/var/lib/dkms/tenstorrent/1.31/build/sg_helpers.h:32: warning: "for_each_sgtable_dma_sg" redefined
 #define for_each_sgtable_dma_sg(sgt, tmp_scl, tmp_idx) for_each_sg((sgt)->sgl, tmp_scl, (sgt)->nents, tmp_idx)
 
In file included from /var/lib/dkms/tenstorrent/1.31/build/sg_helpers.h:4,
                 from /var/lib/dkms/tenstorrent/1.31/build/sg_helpers.c:1:
./include/linux/scatterlist.h:165: note: this is the location of the previous definition
 #define for_each_sgtable_dma_sg(sgt, sg, i) \
 
make[1]: *** [scripts/Makefile.build:317: /var/lib/dkms/tenstorrent/1.31/build/sg_helpers.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from /var/lib/dkms/tenstorrent/1.31/build/memory.c:20:
```

Any pointers would be appreciated.

alewycky-tenstorrent, 2025-03-18:
Lately I've been build-testing with 5.4 and up, so kernels before that need some work. But RHEL backports a lot of changes from later kernels.

dma_(un)map_sgtable and for_each_sgtable_dma_sg weren't in the upstream kernel until 5.8 (+ stable backport to 5.4.233) so we carry replacements. But RHEL 8.4 backports these so we need another special case to use the kernel-provided versions.

For page pinning, the kernel went through many API designs and we have to choose the right one based on the version. RHEL 8.4 backported pin_user_pages_fast and unpin_user_pages_dirty_lock so we can choose those.

I put these changes on the alewycky/rhel-8 branch, please try that out. I had a problem with pci_hw_disabled being multiply-defined, but I suspect that doesn't happen with the build tools in RHEL 8.

Also I could only build-test this branch, I don't have a system to run it on. There could be more problems that I missed.

owainkenwayucl, 2025-03-19:
Thanks - that does indeed install.  Now onto the next hurdle - Rust versions!

owainkenwayucl, 2025-03-19:
Good news!

[image]

milank94, 2025-04-23:
Hey @owainkenwayucl -- do you need any additional support on this issue?

owainkenwayucl, 2025-04-24:
I believe that is all fine - I'm having some other struggles with containerisation but I'll open issues in the appropriate repos when I work out what is going on.

Thanks for your efforts on this!

closed 2025-04-24 by owainkenwayucl