tenstorrent/tt-kmd #83: KMD 2.0.0 introduced regressions with fan control and rpm reporting

bgoelTT, 2025-06-06:
Multiple users on discord are reporting the following behaviour:

After upgrading to the latest tt-kmd 2.0, they sometimes have the problem that the fan runs at full blast all the time. And "fan speed" is shown as 0 RPM
``` 
blackhole-pci-0100
Adapter: PCI adapter
vcore:       740.00 mV 
fan_rpm:        0 RPM
asic_temp:    +44.9°C
power:        34.00 W
current:      47.00 A
```

bgoelTT, 2025-06-06:
One user had 18.4.0, waiting on response from another

joelsmithTT, 2025-06-06:
Thanks @bgoelTT.  Does downgrading to [ttkmd-1.34](https://github.com/tenstorrent/tt-kmd/releases/tag/ttkmd-1.34) restore the expected behavior?  Is the system otherwise functional?  p100 or p150?

@kikkiakki - I tried and failed to replicate the problem on two different systems with p100a cards.

bgoelTT, 2025-06-06:
Okay, I walked through versions with one of the developers, we found KMD==2.0.0 & FW==18.3.0 had the "normal" fan behaviour. However, it still has that short fan "blast" on power-on then settles to ~2000 RPM when idle

closed 2025-07-08 by joelsmithTT

bgoelTT, 2025-07-08:
@joelsmithTT I'm not sure why this was closed as users are still reporting this issue for firmwares 18.4, 18.5, 18.6

Was it closed because this is not a KMD issue and therefore should be opened under https://github.com/tenstorrent/tt-firmware?

joelsmithTT, 2025-07-08:
I closed this issue because it's not a KMD problem.  I don't see a way to transfer it to the tt-firmware project (I can transfer it to other tt projects, but tt-firmware is not in the list).

bgoelTT, 2025-07-08:
Understood, I will open under tt-firmware