tenstorrent/tt-kmd #88: BH: enable ARC watchdog

joelsmithTT, 2025-06-16:
See: https://tenstorrent.atlassian.net/browse/SYS-1494?focusedCommentId=320556
Add this after BH gets ARC messaging support (see: https://github.com/tenstorrent/tt-kmd/issues/53)

tt-rkim, 2025-06-17:
Thank you for filing this

tt-rkim, 2025-07-08:
Any updates on the priority or scheduling of this?

Otherwise, we may have to put a similar hack in CIv2's BH runners.

joelsmithTT, 2025-07-08:
Thanks for the ping on this. https://github.com/tenstorrent/tt-kmd/pull/96

joelsmithTT, 2025-07-08:
It's on `main`.  Do you need me to make a release?

joelsmithTT, 2025-07-09:
I introduced a subtle but serious bug in the ARC messaging implementation, so that is going to have to get fixed before I can make a release.  Apologies in advance for the delay.

> As a side question, "BH ARC messaging support"? This wasn't in the driver until recently? 

wormhole.c knows how to do it for Wormhole, but it wasn't implemented in blackhole.c until recently.  UMD knows how to do it already, independent of KMD.