tenstorrent/tt-kmd #91: Prototype uAPI library

joelsmithTT, 2025-06-19:
# Prototype a User-Space C Library for the KMD Interface

### Summary

Prototype a small C library that exposes the KMD's core functions in an ergonomic way.  This library will serve as a primary interface for all user-space clients, including UMD and diagnostic tools, that need to interact with our WH/BH hardware.

### Goals

* **Near-Term:** Explore the design space by creating a prototype.  Solicit feedback from UMD and tools teams to converge on a stable, well-liked API design.
* **Long-Term:** Provide a documented, versioned, and maintainable reference implementation for all user-space interaction with the KMD.

### Core Requirements & Design Principles

#### 1. Language and Dependencies

* **Language:** Must be written in C, compliant with the C99 standard or later.
* **Dependencies:** The library must have **no third-party dependencies**.  It should only require a standard C Library and standard Linux system headers.

#### 2. API Design

* **Namespace Purity:** The library must not export generic symbols into the global namespace.  All public functions, types, and enums must be prefixed to avoid collisions.  We need to decide on a prefix (e.g. `tt_`, `ttdev_`).
    * *Example:* `tt_device_open()` instead of `DeviceOpen()`.
* **Opaque Handles:** The library must not expose its internal state structures.  All resources must be managed via opaque pointers (handles).  The client will receive a `tt_device_t` from a `tt_device_open` call and will pass it back to subsequent functions.
* **Explicit Error Handling:** The library must provide a clear and robust error reporting mechanism.
* **Statelessness and Thread Safety:**
    * The library itself should be stateless; all state should be associated with a specific handle.
    * All functions must be reentrant and thread-safe.  Concurrent calls with *different* handles must be safe.
    * Operations on a *single* handle from multiple threads at the same time should be explicitly documented as either safe or unsafe (requiring user-side locking).  The latter may be the most practical approach.

#### 3. Functionality

* **Device Lifecycle:** The API must provide explicit functions for discovering, opening, and closing a device.
* **Version Handshaking:** The library must have a mechanism to query the KMD version to ensure compatibility.  The library should refuse to operate if it detects an incompatible driver version.  The library itself should also have a version number that can be queried.
* **Enable host access to device address spaces:** The library must provide a mechanism for the host CPU to directly read and write to the device NOC.
* **Enable device access to the host's memory:** The library must provide a way for the device to perform DMA operations to and from host system RAM.
* **Configure Low-Level NOC Access Parameters:** Accessing the device's NOC is not necessarily a simple read/write; it may involve routing (unicast/multicast) and ordering semantics that must be configurable by the client.

#### 4. Developer Experience

* **Build Simplicity:** The build must work with what LTS Linux distributions ship by default.
* **Documentation:** The public header file must be thoroughly documented using a format compatible with tools like Doxygen.
* **Example Code:** A small `example.c` file demonstrating the library's functionality must be provided.

### Action Items

- [ ] Decide on a permanent namespace prefix.
- [ ] Draft a preliminary header file outlining the proposed API surface.
- [ ] Create a skeleton implementation that maps the proposed functions to the underlying `ioctl` calls.
- [ ] Add a basic Makefile.
- [ ] UMD and Tools teams to review the proposed header and provide feedback.

broskoTT, 2025-06-20:
Is everything (in the initial version) arch agnostic?

joelsmithTT, 2025-06-22:
> everything (in the initial version) arch agnostic?

Yes, I don't think anything specific to one chip type but not another belongs at this level.  There is a question of whether to provide a mechanism for querying architecture specific attributes (e.g. TLB window sizes and quantities).  My thought is no, although some debate about this (and other design decisions) will be healthy.

broskoTT, 2025-06-23:
I guess this needs more discussion/brainstorming. To me it looks like your idea is that there should be two different layers, one would be uApi, then the next one is some "base driver" which shouldn't necessarily live here. Whereas, I don't think there's anything bad from this layer offering up as much common functionality as possible. From the UMD terms, seems like you'd like for this layer to be PCIDevice, whereas I imagine it more at TTDevice layer. @TTDRosen can weight in here on his opinion, but I got a feel he's closer to TTDevice as well.

What's the actual benefit from separating out these two layers? It mostly a set of functions, rather than doing a lot of initialization stuff and keeping stable state etc... 

To be specific, whatever you write as a first version, why would you be against adding something like "wait_arc_init_finish", which is more of a convenience function (knows what to read where), which is optional to call (not called in any form of initializer)? I assume you'd be against since that can be arch specific for example

joelsmithTT, 2025-08-15:
I'm closing this as it is nominally complete, although it's out on a branch and unintegrated.  I think UMD team should decide whether to integrate it or not.  If there is desire to integrate it, I think the code should be re-homed to UMD repo to lower the effort in modifying it (I expect it would have to evolve for UMD to successfully make use of it).